gnea/grbl #1163: Possible bug or compatibility issue grbl v1.1h

boothdcb, 2022-09-20:
I've been experiencing a problem with my new home built CNC router that I suspect might be due to a bug or compatibility issue in grbl v1.1h. 

Setup:
Stepperonline closed loop stepper motors each with CL57T drivers for X,Y and Z axes.
Arduino Uno R3 loaded with grbl v1.1h ( I get similar problems with an Elegoo Uno R3 - arduino clone)
Software: UGS Platform

Issue: The number of pulses received by the stepper motor drivers for each and every jog command is less than that required to move the stepper motors the required amount. The magnitude of the error appears constant, regardless of the distance of the move. The error seems to vary with the feed rate (although, strangely, not the acceleration) but it does appear constant for any given feed rate. When executing a gcode file in UGS Platform, the reduced number of pulses received by the stepper motor drivers affects the very first movement command in the gcode file but does not apply to subsequent commands in the file. This results in the physical work volume being displaced by a set amount each time a gcode file is run. This displacement is cumulative if a job requires tool changes and separate gcode files for subsequent operations.

langwadt, 2022-09-20:
are you disabling the stepperdrivers when idle?

boothdcb, 2022-09-21:
I'm not aware that they are being disabled. I have not intentionally disabled them. They would have to be automatically disabled after every jog command and at the end of any streamed gcode file by UGS or grbl. I'm not sure how I would test for this or prevent it, if that is the cause.

boothdcb, 2022-09-21:
Thank you for responding to my post.
I'm not aware that they are being disabled. I have not intentionally disabled them. They would have to be automatically disabled after every jog command and at the end of any streamed gcode file by UGS or grbl. I'm not sure how I would test for this or prevent it, if that is the cause.
Best regardsDavid

langwadt, 2022-09-21:
do you have theena+/- connected? if so try disconnecting them on the drivers, that will keep them enabled always

boothdcb, 2022-09-21:
Thanks, the ena+/- is connected so I will try disconnecting them.

boothdcb, 2022-09-22:
Eureka!! Thank you so much langwadt for that vital suggestion. The problem that I've been tearing my hair out with for the last 3 weeks is now solved! The router now behaves perfectly. So much for the manufacturer's wiring diagram and all their other unhelpful possible causes and solutions. These things should come with a health warning.

langwadt, 2022-09-22:
I'd say it isn't entirely the manufacturers fault, grbl apperently just doesn't take into account that there might be a delay from enabling drivers until they are ready

terjeio, 2022-09-25:
Enabling a [STEP_PULSE_DELAY](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/config.h#L416-L425) could be a workaround if you want to disable motors when not moving. And setting `$1=255` should keep them on at all times.