gnea/grbl #1170: Unintended delay on turning on spindle automatically

Manttech, 2022-10-19:
I have a home-made CNC running on an Arduino UNO with GRBL and UGS as interface.
Recently I installed a relay for turning on and off automatically the spindle. The spindle its a woodworking Bosch router, no variable speed.
In the config.h I commented (disabled) "#define VARIABLE_SPINDLE".
The issue is that it seems to be a fixed time delay (around 2-3 seconds) from the moment I press play to the turning on of the spindle, but the machine (steppers) runs with no delay and the tool digs into the work before spinning.
The workaround I found is to jog the machine far away from the starting position, so it will take 2-3 seconds to get to work and give time for the spindle to turn on.
Of course, this is far from ideal and annoying and I would like to fix it.
I tried the relay with a 5v battery in an 220v old welder before installing on the CNC and wasn't any noticeable delay.

I used Fusion 360 as a CAM and a typical file looks like this:

(FileName)
(T1  D=3.175 CR=0 - ZMIN=0 - Toolname)
G90 G94
G17
G21
(When using Fusion 360 for Personal Use, the feedrate of)
(rapid moves is reduced to match the feedrate of cutting)
(moves, which can increase machining time. Unrestricted rapid)
(moves are available with a Fusion 360 Subscription.)
G28 G91 Z0
G90

(Toolpathname)
M3 S25000
T1
G54
M8
G0 X20.738 Y-15.122
Z25
G1 Z15 F2000
Z14.5 F1000
Z6.984
X20.74 Y-15.114 Z6.914 F2000
X20.744 Y-15.091 Z6.846
X20.75 Y-15.054 Z6.786
X20.759 Y-15.005 Z6.736
(....)
(rest of the file)

Any ideas? thanks!

mar0x, 2022-10-19:
Hello,

Consider using "G4 P**xx**" command to pause CNC for **xx** seconds and wait for spindle spin up.

langwadt, 2022-10-19:
you can edit the post processor in fusion so it adds a dwell after turning on spindle

Manttech, 2022-10-22:
Works fine on any other G-code sender, seems to be that UGS it's just too buggy.

closed 2022-10-22 by Manttech