gnea/grbl #1174: grbl doesn't fit on atmega328p, master version, didn't change anything

andrewboktor, 2022-11-14:
The hex file I'm getting without modifying anything won't fit on atmega328p.
I'm really stumped here. The released binary from 2019-08-25 do fit but I just need to swap some pins!
Output of build and avr-gcc --version below
```
[andrew@flex5 grbl]$ make
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/motion_control.c -o build/motion_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/gcode.c -o build/gcode.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/spindle_control.c -o build/spindle_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/coolant_control.c -o build/coolant_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/serial.c -o build/serial.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/protocol.c -o build/protocol.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/stepper.c -o build/stepper.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/settings.c -o build/settings.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/planner.c -o build/planner.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/nuts_bolts.c -o build/nuts_bolts.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/limits.c -o build/limits.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/jog.c -o build/jog.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/print.c -o build/print.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/probe.c -o build/probe.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/report.c -o build/report.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -MMD -MP -c grbl/system.c -o build/system.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -flto -o build/main.elf build/main.o build/motion_control.o build/gcode.o build/spindle_control.o build/coolant_control.o build/serial.o build/protocol.o build/stepper.o build/eeprom.o build/settings.o build/planner.o build/nuts_bolts.o build/limits.o build/jog.o build/print.o build/probe.o build/report.o build/system.o -lm -Wl,--gc-sections
rm -f grbl.hex
avr-objcopy -j .text -j .data -O ihex build/main.elf grbl.hex
avr-size --format=berkeley build/main.elf
   text    data     bss     dec     hex filename
  36512       0    1633   38145    9501 build/main.elf
[andrew@flex5 grbl]$ avr-gcc --version
avr-gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

dungjk, 2022-11-15:
GRBL is quite tight fit to atmega328p.
You're using new version of avr-gcc, it may add new features and increase size of hex.
Some suggest solution:
- Try to compile with release mode and higher optimization.
- Use older version of avr-gcc
- Disable some features of GRBL that not use.

Andy1978, 2024-07-30:
I can confirm this when compiling on MSYS2:

```
mingw-w64-ucrt-x86_64-avr-gcc-12.2.0-1
avr-size --format=berkeley build/main.elf
   text    data     bss     dec     hex filename
  36512       0    1633   38145    9501 build/main.elf

mingw-w64-ucrt-x86_64-avr-gcc-8.5.0-1-any.pkg.tar.zst
avr-size --format=berkeley build/main.elf
   text    data     bss     dec     hex filename
  29818       0    1633   31451    7adb build/main.elf
```

So 8.5.0 is fine, 12.2 not