gnea/grbl #1177: Dual Z-Axis support?

richardklingler, 2022-11-24:
Good morning

When going through the issues here I only see that dual axis is only supported on X and Y.
Though still don't know how to configure those (o;

I have an Anycubic 3d printer here with Mega2560 microcontroller where I want to put Grbl on,
but it has dual Z-axis with endstops at each side.

I don't see any option with Grbl how to achieve that...or if it is even possible.

GrahamJB, 2022-11-24:
For a 3D printer you need to be looking at Marlin, not Grbl.

GrahamJB, 2022-11-24:
> The Anycubic will be fitted with an LED laser :-)
>
> So in the end nothing to do with 3d printing as I will use Lightburn to control the machine...

OK, that sounds good.

But unfortunately I cannot help.

melvinisken, 2022-11-25:
> The Anycubic will be fitted with an LED laser :-)
> 
> So in the end nothing to do with 3d printing as I will use Lightburn to control the machine...

That's not a problem. Marlin supports being used for a laser, and is also supported by LightBurn. You don't need to use grbl necessarily. If you don't want to change your board, I suggest looking into how to use Marlin with a laser and LightBurn.