gnea/grbl #1178: G2/G3 No axis words in plane (too strict rule)

kimstik, 2022-11-28:
It seems that GRBL require at least one in-plane axis word for G2/G3.

In case of arcs is correct. But in case of full circle/helix it may fail.
There is correct gcode which fail with code 32 here:
`if (!(axis_words & (bit(axis_0)|bit(axis_1)))) { FAIL(STATUS_GCODE_NO_AXIS_WORDS_IN_PLANE); } // [No axis words in plane]` 

I think the rule is too strict.

```
G1 Z0 F500
G17
G2 X7 Y0 Z-1 I-7 J0
Y0 Z-1 I-7 J0 (ok)
Z-2 I-7 J0 (fail)
```

The other issue is that it didn't stop the movement on the error (in my case). I want to understand why...

terjeio, 2022-11-30:
> The other issue is that it didn't stop the movement on the error (in my case). I want to understand why...

The sender user character counting mode for transmission? If so Grbl wil continue executing any buffered gcode unlike grblHAL that repeats the error and ignore gcode until error is cleared...

kimstik, 2022-11-30:
I used Candle as sender.