gnea/grbl #1185: height map for PCB

Zdenekr, 2022-12-22:
Hello,
i use candle to make pcb and i started to make pcb with metallized holes.
First I drill all the holes, do the plating and then I need to mill the paths.
For that I need a height map.
Sometimes I hit a hole while scanning and the map is corrupted.
Is there any way to scan the bond paths outside the holes?
Thank you for answer.
Regards
Zdenek

GrahamJB, 2022-12-23:
I would try a probe with a flat base of a larger diameter than the holes first and see how that goes.

 

Just an idea, not necessarily a good one.

Zdenekr, 2022-12-23:
thanks for the tip, i'll try it

closed 2022-12-23 by Zdenekr

pmbrown141, 2023-01-07:
Zdenekr

I would love to know your process for plating through holes. 

Would you care to share?