gnea/grbl #1186: xLoader instructions lead to a trojan

skruger, 2022-12-25:
I followed the windows instructions to load a grbl binary on to my arduino using xLoader.
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino#for-windows

The xLoader didn't actually work and windows identified a Trojan and named xLoader as being related to it's installation.

[image]

AlexHolden, 2022-12-25:
I've deleted the link, just in case. I've never used xLoader myself so I'm not sure what is the official site to download it from.

X3msnake, 2022-12-25:
XLoader is a sourceless wrapper to avrdude and since the author never released the ssourcecode and his website disappered the copies arount the web can not be thrusted

i reccomend people use avrdudess gui instead it is opensource and updated. and donwload only from the source GitHub

https://github.com/ZakKemble/AVRDUDESS

harryberlin, 2022-12-25:
BTW:
Laser GRBL has a Dialog for uploading Firmware.

arrsoft312, 2022-12-27:
You can give a try to my small and simple utility for flashing HEX-files to Arduino UNO and Nano boards.

Source code: https://github.com/arrsoft312/update
Compiled binary for Windows: https://github.com/arrsoft312/update/releases/latest

skruger, 2023-01-02:
I followed the CLI instructions so I was able to get the image flashed. I just wanted to make sure this was reported to raise awareness. Thanks for all your suggestions, I may incorporate them when I have more arduinos to flash.

closed 2023-01-02 by skruger

per1234, 2023-03-23:
>I'm not sure what is the official site to download it from.

This is the XLoader download link that was in the wiki before the chain of malicious edits:

http://www.hobbytronics.co.uk/download/XLoader.zip

You can see it in the wiki commit history here:

https://github.com/gnea/grbl/wiki/_compare/bf8da25354e311faaafc15ffae4819c59b3b7b2f

---

Related: https://github.com/gnea/grbl/issues/1053