gnea/grbl #1188: Noob questions

dragonflz, 2023-01-19:
Help! I think I messed up my laser. 
I'm brand new to all of this and I need a lot of help. I have an Atom Stack X7 pro. 
I was able to input commands on grbl but something happened that I don't know how to fix.
In the grbl window it constantly cycles through over and over. I'm not able to enter any commands not even help. I can't even get my configure grbl to do anything. It tells me there was an error. I am enclosing the session log for review. If anyone can help me I would greatly appreciate it.
Thank you!
8/01/2023 13:17:05.175	Program     	------- LaserGRBL v4.9.4 [64bit] START -------
18/01/2023 13:17:08.973	Update DB   	Material DB is up to date
18/01/2023 13:17:23.730	SetStatus   	Machine status [Connecting]
18/01/2023 13:17:23.772	OpenCom     	Open COM3 @ 115200 baud Ctrl-X (UsbSerial)
18/01/2023 13:17:23.775	OpenCom     	Error: Access to the port 'COM3' is denied.
18/01/2023 13:17:23.775	SetStatus   	Machine status [Disconnected]
18/01/2023 13:17:40.284	SetStatus   	Machine status [Connecting]
18/01/2023 13:17:40.285	OpenCom     	Open COM3 @ 115200 baud Ctrl-X (UsbSerial)
18/01/2023 13:17:40.285	OpenCom     	Error: Access to the port 'COM3' is denied.
18/01/2023 13:17:40.285	SetStatus   	Machine status [Disconnected]
18/01/2023 13:17:56.188	SetStatus   	Machine status [Connecting]
18/01/2023 13:17:56.189	OpenCom     	Open COM3 @ 115200 baud Ctrl-X (UsbSerial)
18/01/2023 13:17:56.227	SendImmediate	Send Immediate Command [0x18]
18/01/2023 13:18:02.751	SetStatus   	Machine status [Idle]
18/01/2023 13:18:13.323	Refresh Config	System.TimeoutException: No response received from grbl!
18/01/2023 13:18:13.323	Refresh Config	   at LaserGRBL.GrblCore.RefreshConfig() in E:\LaserGRBL\LaserGRBL\Core\GrblCore.cs:line 1003
18/01/2023 13:18:54.226	SendImmediate	Send Immediate Command [0x18]
18/01/2023 13:22:55.265	SendImmediate	Send Immediate Command [0x18]
18/01/2023 13:25:36.091	OpenFile    	Open C:\Users\drago\OneDrive\Desktop\kitchen sayings\coffee.png
18/01/2023 13:26:51.914	CloseCom    	Close COM3 [USER]
18/01/2023 13:26:51.943	SetStatus   	Machine status [Disconnected]
18/01/2023 13:26:58.221	Program     	-------------

Dio3L, 2023-02-02:
Which application do you use to control your CNC ?
Do you use the COM3 with an other software, like arduino with the serial monitor ?
if this is the case, it is necessary to close the app or the serial monitor who use your COM3

dragonflz, 2023-02-02:
Hello,
I use com3 with Lasergrbl if that helps. I am able to connect to
Lasergrbl but I get stuck when trying to send commands to grbl. I can't
upload any changes. The console where the commands are just keeps moving
through the same few lines. I'm sure there are technical terms for what I'm
trying to say but I don't know what it is. I can't flash the firmware, when
I try to bring up grbl configuration I get an error message. I'm very
frustrated so any help you offer would be greatly appreciated. Thank You!!

RCSrohan, 2023-02-02:
Do you use a windows system, if so 

Step 1 - Frist try unplugging your usb for your CNC and plug it into a different socket on your computer. 

This should assign it a new communication port of which you can select from the drop-down list within laser grbl. 

Step 2 - if that does not work.

Click start from your Windows computer and type device manager.

When device manager opens qlikview and then show hidden devices from the dropdown menu once you have clicked show hidden devices find horse com' and lpt click on the arrow to the left hand side of ports.

This will show you all of the communication ports that are plugged in to your computer.

It should show you that you have com3 or your designated communication port for your CNC, 

It should also show you that it is a USB serial ch340.

To manually change the com' port number you can right click on that comport left click on properties left click on port settings left click on advanced and under.com port number click the dropdown menu and select a number that is not in use. 

Once you have selected a new com port number, 

You would go back to lasergrbl and select that comport from the drop down menu.

Hope that helps.

dragonflz, 2023-02-02:
Thank you for your help. I will do that🙂