gnea/grbl #1191: Grbl don't stop motors after end stop trigered

Dio3L, 2023-02-02:
**Hi, here is my problem : my limit switches are well interpreted by the arduino but it won't stop the motors and we can control it in both directions.**

I try to renovate CNC (change old electronic cards) and use it with arduino, [tb6600](https://www.google.com/url?sa=i&url=https%3A%2F%2Fwww.ebay.fr%2Fitm%2F173516178540&psig=AOvVaw2NxHav39ixZ0kqaY5dj_Ow&ust=1675416176640000&source=images&cd=vfe&ved=0CBAQjRxqFwoTCNCaxOLB9vwCFQAAAAAdAAAAABAI) stepper motor driver and grbl 1.1h version.
I use [bCNC](https://github.com/vlachoudis/bCNC) to control grbl. 

My XYZ limit switches are NC so the limit pins are connected directly to 5V and when the switch is trigered, limit pin receive a 0V.
I have checked the wiring several times and it is correct, I've tested different configurations and different versions of grbl 1.1 and 0.9 but it never get worked : I still move my motors in both directions when limit switch activated, in each axis.
Someone can help me 

There is my configuration : 

$0=10
$1=25
$2=1
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=79.000
$101=79.000
$102=79.000
$110=800.000
$111=800.000
$112=800.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

**here, bCNC tell to the right of the 'idle mode' (grbl wait instructions) which switch is activated or not :**
[image]

**here, just the X limit switch is triggered, Y and Z are free :**
[image]

If someone have an idea i can't solve it.

Dio3L, 2023-02-02:
> $21=0, that means hard limits are not enabled

The hard limit doesn't work either.
I tried with and without hard limit or soft limit
I re-upload grbl many times: I tested the correct config and with the default config.h and the default serial configurations but also with differents serial configurations.
grbl doesn't stop motors movement when limit pin activated (or disabled if i have a wrong config with $5=0) but it still detects them correctly,.

vlastikcocek, 2023-02-17:
do you have a shielded cable for the limit switch?

Dio3L, 2023-05-15:
> do you have a shielded cable for the limit switch?

I don't have shielded cable for the limit switch, i wired RC filters to reduce the noise
but i think i don't need it because arduino detect the two state, high or low correctly and he don't stop the motors anyway.