gnea/grbl #1192: Grbl 1.1f ['$' for help]

Yasir898, 2023-02-09:
i face a problem 
pen plotter is working fine like stepper motor and servo 
but  after a some execution it stack and pause and show the error of grbl 1.1f and i have to do it again
[image]

Yasir898, 2023-02-09:
i face a problem
pen plotter is working fine like stepper motor and servo
but after a some execution it stack and pause and show the error of grbl 1.1f and i have to do it again

**** Connected to COM3 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> G21G90 G0Z5
>>> G90 G0 X0 Y0
>>> G90 G0 Z0
ok
ok
ok
>>> $J=G21G91X10Y10F10000
ok
**** Connected to COM3 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
[... 538 lines cut ...]
>>> G1X73.0957Y77.4824
>>> G4P0.2
>>> M05S0
>>> G4P0.2
>>> G1F10000
>>> G2X68.6787Y80.2104I1.6554J7.6204
ok
ok
ok
>>> G2X66.976Y85.0781I5.9919J4.8276
ok
ok
ok
>>> G2X67.5855Y88.1355I8.187J-0.0428
ok
Grbl 1.1f ['$' for help]

**** Canceling file transfer. ****

>>> $J=G21G91X10Y10F10000
ok
errr:7errr:7
An unexpected error was detected: 

[image]

vorlovsky, 2023-07-06:
It looks like a connection issue. To find the "bottle neck" I should advise you first to run the same gcode file with the CNC shield disconnected and to use different USB cables. In case nothing changes and if your Arduino is a clone of Uno board with CH340 USB bridge, try to replace it with an official Uno board that has ATmega bridge.