gnea/grbl #1196: Zprobe issue

ReinierGielen, 2023-02-19:
Hello I have a DIY cnc with an arduino UNO and Grbl 0.9f and it works fine.
We used a makeshift Z-probe connected to GND and A5 and that worked.
No I was smartening up the build and ordered a standard z-probe from AliExpress.

Now I am having issues with the Z-probe. I checked the wiring and fixed an issue.
I now have 4,9v difference on the probe and when I manually touche the clamp on the touchplate during probing the machine responds like expected.
BUT when I connect the clamp to the drill bit and start a probe it goed up and not down.
So I measured the voltage between the drill and the touchplate and found 3.1v
I assume this is to low and makes GRBL think the drill is touching the plate.

2 questions. 
- why am I losing olmost 2v when connected to the drill ?
- how kan I fix this ? Hardware changes ? trigger level in the software ?

Any help is appriciated

GrahamJB, 2023-02-19:
The movement of the Z probe is controlled by your Gcode sender software (Candle, UGS, Openbuilds etc.) NOT Grbl!

 

Check the settings your sender program uses to perform the Z probe sequence.

ReinierGielen, 2023-02-19:
I am using UGS 0.9f and the gcode is G38.2 Z-10 F100 which I also tried manually from the console.
As I explained the code does what it is supposed to doe and responds correctly when I touch the touchplate with the clamp. The connection is detected and the direction is reversed.

But this does not work when the clamp is attached to the drill bit. The the drill goes up not down.

The only differens I could fine was the voltage difference

GrahamJB, 2023-02-19:
Is your movement mode absolute (G90) or relative (G91)?

 

The Gcode I use for Z probing is 

G91; G38.2 Z-100 F100; G0Z1; G38.2 Z-2 F10; G92 Z19.10; G0 Z2

 

My probe base is 19.10mm thick.

ReinierGielen, 2023-02-19:
Probably the same probe as mine 19,16mm. I use the same commands. The actual probe command is G38.2 it moves the drill down 100 ( Z-100 ) with the speed 100 ( F100 ). It is all doing that when the probe is NOT connected to the drill. As it should because the pin A5 with 4.9v is HIGH. As soon as the probe loop is closed the pin drops to 0 and the movement stops and retracts.

But if the probe is connected to the drill it retracts directly when the command is given. This is what is should do is it is already touchin the plate making A5 LOW. Therefore my question is 3.1v considered to be low ?

GrahamJB, 2023-02-19:
My apologies,

I now fully understand what you are saying.

 

I would check the power wiring and earthing of the router board. Connecting the probe base to the bit by a clip should not give that voltage drop.

Is 3.1V too low, I don’t know, I would have to check the Arduino specs but I think the problem lies elsewhere.

langwadt, 2023-02-19:
3.1V is on the edge for 5V.  What connects to the bit, GND or A5?  if it is A5 try swapping the connection so the bit is GND

ReinierGielen, 2023-02-19:
LOL, that was just what I was doing now and it indeed solved the problem. I was color blind. The probe has a red and black wire. The plate is black the clamp is red. And ofcourse I connecte black to ground and red to power. But in this case that is not the right way to connect it. On my previous probe the wires where reversed and it always worked.
This way there is no voltage drop at the start and that explains everything.

Thanks for all the help