gnea/grbl #1198: PWM output inversion option

Cyrille-de-Brebisson, 2023-02-22:
Hello,

Is there any easy way to invert the PWM output for active low devices? I wanted to control a K40 laser directly from the grbl board, but it looks like the K40 is active low, not high, so everything is inverted...

There is a way to invert lots of things iwth grbl (directions, limit switches), but I have not found a way to invert the PWM...

Unless I can switch $30 and $31 so that $30 (spindle max is 0) and $31 (spindle min is 1000)...

Thanks again for all the work you put in grbl guys!
Cyrille

langwadt, 2023-02-22:
Seems like it would be a safety issue, since default/boot/off would be full power on the laser. 
Just add a transistor (or optocoupler) to do  the inversion.

Cyrille-de-Brebisson, 2023-02-23:
Hello, I guess you are right (darn it, I feel bod for not niticing that this was a problem)...
I wanted to avoid having to add extra HW to a "standard" board...
Thanks for the reality check.

Cyrille-de-Brebisson, 2023-02-23:
Hello, I guess you are right (darn it, I feel bod for not niticing that this was a problem)...
I wanted to avoid having to add extra HW to a "standard" board...

Thanks for the reality check.

I will see if there is a way to handle things differently since the K40 power supply has a separate PWM and Fire pins, they might still be a way to do it... I wonder how it reacts if I tell it "Fire" with PWM set to 0 for example. If it does not energize the laser, then it might be a sollution.

Regards,
Cyrille

Cyrille-de-Brebisson, 2023-02-26:
Hello,

Well, the arduino will be powered by the laser power supply, so it is not too big of an issue.
Also, the laser PWM will be linked to the arduino PWM, so even at startup time, yes, the laser will be "ON" until the init of the enable pin, BUT, the PWM will be at 0, so the laser should be on, with a power of 0, so I do not think that it will be an issue...

If I have to go the HW route, I will, but if I can avoid it... better...

Thanks for all your help!
Cyrille

Cyrille-de-Brebisson, 2023-03-14:
Hello,

In case someone wants to do the same thing as I did. It is working!
The steps are:

    Modify GRBL to invert the spindle enable pin (INVERT_SPINDLE_ENABLE_PIN)
    in my case, I remapped the spindle enable to the Z limit (SPINDLE_ENABLE_BIT) (and the Z limit to the same pin as Y limit). As my board did NOT have a wired spindle enable pin available.
    added a call to spindle_init(); at the top of main to make sure that it would be driven high ASAP at boot time.
    added some wires (male JST VH 3.96mm for the power and jst xh 2,54 for the signal (laser ON and PWM))

No need for HW inverter for the laser!

Thanks to the comunity for your help!
Cyrille