gnea/grbl #1199: Circle cut in half.

LvWDF, 2023-02-23:
Hi
Arduino uno, grbl an ugs.
When I try cutting a circle it starts cutting clockwise until 180' and then comes straight down to the starting point. (Half e circle)

Already tried different cam programs to create gcode. 3 different arduino's. Open builds in stead of ugs. Switching x an y axle.  But result is always the same.

Cnc does make a perfect square with correct measurements.

Who can help?
Kind regards
Ben

neilferreri, 2023-02-23:
That's not a Grbl issue.
What kind of machine?

MARIOBASZ, 2023-02-23:
can you upload the example? it could be a code error.

LvWDF, 2023-02-24:
G21 ; millimeters
G90 ; absolute coordinate
G17 ; XY plane
G94 ; units per minute feed rate mode
M3 S1000 ; Turning on spindle

; Ellipse - On path (1/2)
G0 Z5
G0 X38.5Y11
G0 Z0
G1 F400 X38.5Y11Z-0
G1 F1000 X38.5Y11Z-0
G1 X35.688Y11.142Z-0
G1 X32.958Y11.559Z-0
G1 X30.322Y12.236Z-0
G1 X27.796Y13.161Z-0
G1 X25.392Y14.319Z-0
G1 X23.124Y15.697Z-0
G1 X21.007Y17.28Z-0
G1 X19.055Y19.055Z-0
G1 X17.28Y21.007Z-0
G1 X15.697Y23.124Z-0
G1 X14.319Y25.392Z-0
G1 X13.161Y27.796Z-0
G1 X12.236Y30.322Z-0
G1 X11.559Y32.958Z-0
G1 X11.142Y35.688Z-0
G1 X11Y38.5Z-0
G1 X11.142Y41.312Z-0
G1 X11.559Y44.042Z-0
G1 X12.236Y46.678Z-0
G1 X13.161Y49.204Z-0
G1 X14.319Y51.608Z-0
G1 X15.697Y53.876Z-0
G1 X17.28Y55.993Z-0
G1 X19.055Y57.945Z-0
G1 X21.007Y59.72Z-0
G1 X23.124Y61.303Z-0
G1 X25.392Y62.681Z-0
G1 X27.796Y63.839Z-0
G1 X30.322Y64.764Z-0
G1 X32.958Y65.441Z-0
G1 X35.688Y65.858Z-0
G1 X38.5Y66Z-0
G1 X41.312Y65.858Z-0
G1 X44.042Y65.441Z-0
G1 X46.678Y64.764Z-0
G1 X49.204Y63.839Z-0
G1 X51.608Y62.681Z-0
G1 X53.876Y61.303Z-0
G1 X55.993Y59.72Z-0
G1 X57.945Y57.945Z-0
G1 X59.72Y55.993Z-0
G1 X61.303Y53.876Z-0
G1 X62.681Y51.608Z-0
G1 X63.839Y49.204Z-0
G1 X64.764Y46.678Z-0
G1 X65.441Y44.042Z-0
G1 X65.858Y41.312Z-0
G1 X66Y38.5Z-0
G1 X65.858Y35.688Z-0
G1 X65.441Y32.958Z-0
G1 X64.764Y30.322Z-0
G1 X63.839Y27.796Z-0
G1 X62.681Y25.392Z-0
G1 X61.303Y23.124Z-0
G1 X59.72Y21.007Z-0
G1 X57.945Y19.055Z-0
G1 X55.993Y17.28Z-0
G1 X53.876Y15.697Z-0
G1 X51.608Y14.319Z-0
G1 X49.204Y13.161Z-0
G1 X46.678Y12.236Z-0
G1 X44.042Y11.559Z-0
G1 X41.312Y11.142Z-0
G1 X38.5Y11Z-0
G0 Z5
; Ellipse - None (2/2)

; Turning off spindle
M5

LvWDF, 2023-02-24:
Its a cnc plasma cutter. arduino+ grbl based.

same issue when plasma is connected or not... cables are shielded.

SailWithChips, 2023-02-24:
Could you add your grbl settings?.

have you checked to start your program (gcode) on other place?, to check if its a hardware or softlimit for instance.

LvWDF, 2023-02-26:
Hi

Today I reinstalled all the software concerning the cnc and  that seamed to have solved the problem. Still don't know what went wrong.

Thans for the help!
Kind regards 
Ben

closed 2023-02-26 by LvWDF