gnea/grbl #1200: Gtbl v1.1 report request "?" only responds with "ok"

FaysalAriss, 2023-03-01:
I've tried using this command on UGS and Candle, I type in "?" no carriage return or spaces or anything and send it and the only response is ok.
[image]

psyhlo, 2023-03-01:
type $ for help
it is written on the line 3

FaysalAriss, 2023-03-01:
[image]
"$" just gives me a list of commands I'm able to use, I've reviewed all of them but they don't fix my problem.

FaysalAriss, 2023-03-01:
The output is supposed to be hidden to the user, to enable verbose output in UGS: Tools>Settings>UGS>Sender Options>Show verbose output

closed 2023-03-01 by FaysalAriss