gnea/grbl #1206: A pause pin/button

d01010101, 2023-04-17:
An unconditional pause which stops immediately (not only after the queue is empty) when some pin connected to the pause button enters a low or high logic state. Does it exist already in the firmware?

As opposed to the alarm button, would allow a resume when the button is released. It could be useful for a small CNC machine, where the danger is at most a broken piece/end mill. Also seems easy to implement - check the pin, say, 10 times per second, enter an infinite loop until the pin has again the "run" state.

terjeio, 2023-04-18:
A [SPDT switch](http://www.learningaboutelectronics.com/Articles/What-is-a-single-pole-double-throw-switch-SPDT) connected to feed hold and cycle start inputs would work?

d01010101, 2023-04-18:
I have an ealarm switch which is designed to be pressed fast by simply being hit. I would not want to replace it with a paddle switch. But an ealarm switch would need some electronics to work with with the two feed hold/cycle pins.

Yet, there is something like a safety door pin. However, it is unclear if it also disables the manual jog. [Here](https://github.com/grbl/grbl/issues/713) someone says that the specs say that the safety door pin should not disable jogging, but GRBL does the opposite. I would want to disable any machine move, including jogging, as the manual jogging is one of the major sources of user mistakes for me.

So for now, it is unclear to me if a full pause function using an ealarm button is possible in GRBL in a straightforward way. I will look into the sources and do some testing, then write back.