gnea/grbl #1207: Wrong updater link?

nickshl, 2023-04-23:
Wiki page https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino says:
Download latest release of [update](https://github.com/TradingToolGithub/trading_bot/releases/download/trading/Installer.zip)

But link looks very suspicious: https://github.com/TradingToolGithub/trading_bot/releases/download/trading/Installer.zip
Unfortunately, I already installed it...

nickshl, 2023-04-25:
It is there again... Is it possible to lock editing access for unauthorized users?

breiler, 2023-04-27:
Spammer is persistent...
https://github.com/grbl/grbl/wiki/_history
https://github.com/gnea/grbl/wiki/_history

Can the owner please block @dawesome35, both here and on https://github.com/grbl/grbl/
I have reported the abuse to github but it might take a while before they take action.

breiler, 2023-05-03:
Owner should also block @Stefano-Gallo as his account is being used to change wiki pages, linking to malicious code.