gnea/grbl #1209: It needs to be configured mannualy

Yonaskassa, 2023-04-27:
I am trying to connect my arduino mega 2560 via ugs but it says
"Connected to Grbl
The setup wizard does not know how to configure this type of controller.
It needs to be configured manually."
please any one can help me?

breiler, 2023-04-27:
Exactly which version of GRBL did you install and how did you install it?
The reason I ask is that you have posted the issue on `gnea/grbl` which doesn't support the Mega 2560. You are perhaps using the `https://github.com/gnea/grbl-Mega` or ´https://github.com/fra589/grbl-Mega-5X`?

Have you tried verifying that the firmware was properly installed, you can easily do this using the `Arduino IDE Serial Monitor`. Connect to the serial port with baud 115200 and make sure that you get a Grbl welcome message.

Make sure that you are using the latest version of UGS 2.0.17, it has some protocol changes specifically for initiating communication with Mega 2560 boards: 
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.17

Yonaskassa, 2023-04-27:
Thanks for your reply, I am trying by grbl 1.1 and ugs v2.0.14.
Let me try as you suggest and back to you soon.

Yonaskassa, 2023-04-28:
Dear,
Thanks so much its working know!🙏

etempm, 2023-04-28:
Hi, I have same issue with my uno.
I am trying to connect arduino UNO TO ugs but it says
"Connected to Grbl
The setup wizard does not know how to configure this type of controller.
It needs to be configured manually."

grbl VERSİON İS 1.1h and UGS version is 2.0.17.

Any suggestion?

breiler, 2023-04-28:
@etempm and you are certain that the firmare is loaded to the controller properly?

> Have you tried verifying that the firmware was properly installed, you can easily do this using the Arduino IDE Serial Monitor. Connect to the serial port with baud 115200 and make sure that you get a Grbl welcome message.

etempm, 2023-04-28:
Yes. I tried 2.0.15 and 2.0.17 with no luck. But now I tried with UGS 2.0.14 and it works. It seems that, problem with version 2.0.15 and upper.