gnea/grbl #1210: Path

Yonaskassa, 2023-04-28:
Hi Dear,
[image]
As you have seen a design on UGS the path is not random moving, Cnc moves unnecessary fwd and bwd movment. How to make the path moving from the nearest to other side.
Sorry for my english.

breiler, 2023-04-29:
The picture wasn't uploaded properly

Yonaskassa, 2023-04-29:
[image]

breiler, 2023-04-29:
This is not an issue with GRBL, you would be better served by using the UGS discussions: https://github.com/winder/Universal-G-Code-Sender

In the latest version of UGS I removed the automatic ordering of objects as a request from another user.

Each geometric object will be processed in the order they are listed in the entities list. However if you have one complex geometric path object with a lot of holes the ordering of how each hole is processed could probably be optimized. Did you import an SVG, could you attach a sample file?

Yonaskassa, 2023-04-29:
Sure, Here is attached sample .svg file.
Regards!
[image]

breiler, 2023-05-02:
Yeah, this should be more optimized. I have created this issue in UGS: https://github.com/winder/Universal-G-Code-Sender/issues/2209

breiler, 2023-05-03:
There is a new nightly build available that should optimize this a bit:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/nightly