gnea/grbl #1211: Frustration with GRBL and IDE

Cabinfourus, 2023-05-02:
I am not able to load grbl 1.1h into IDE. I have tried adding it as a zip and I have tried manually putting it in the libraries folder in documents./Arduino. NOTHING works.

Can anyone tell me what the heck is going on?
Marlin isn't this difficult.

mar0x, 2023-05-03:
Hello.
Please follow this instruction steps: https://github.com/gnea/grbl/wiki/Compiling-Grbl . If anything goes wrong, share the details: complete information about your setup (OS version, Arduino IDE version etc.), the step which is failed, the messages/errors you've got vs. your expectations.

pawlomik2, 2023-05-12:
> I am not able to load grbl 1.1h into IDE. I have tried adding it as a zip and I have tried manually putting it in the libraries folder in documents./Arduino. NOTHING works.
> 
> 
> 
> Can anyone tell me what the heck is going on?
> 
> Marlin isn't this difficult. 

You can try to download winrar, you don't need buy it just unpack the zip by it