gnea/grbl #1216: Error 22 while probing heightmap

oarcher, 2023-06-11:
When probing heightmap, I got the following errors:
```
Error message(s) received:
3: G38.2Z-2 < error:22
6: G38.2Z-2 < error:22
9: G38.2Z-2 < error:22
12: G38.2Z-2 < error:22
```

My Zt is 2.0, and Zb is -2.0. If I enter `G38.2Z-2`, I also got the error 22.

GrahamJB, 2023-06-11:
Error 22 is no federate has been specified, how fast are you going to be probing?

oarcher, 2023-06-11:
I'm sorry, I've post this issue in the wrong place: It's a Candle related issue.

closed 2023-06-11 by oarcher

GrahamJB, 2023-06-11:
I would say it’s not candle related, Grbl is returning the error.

 

Earlier in your Gcode is there an Fnnnn line? It’s safer to add one to each G32.2 line like G38.2 Z2 F100, that will fix it.

oarcher, 2023-06-11:
The Candle default setting for z-probe feed is zero, and that not generate the Fxxx, causing the error. Setting it to 100 do the tricks.

Thanks!