gnea/grbl #1222: Limit switch alarm triggered also when releasing

SebastianKusykIFJ, 2023-07-18:
Hello,
I hit a limit switch, got error info from GRBL and resetted and unlocked it. However, when I started moving in the opposite direction, I also got the limit switch error when the switch was released and had to reset and unlock again. How to prevent this? Is it caused by switch bouncing / electrical oscillations or by GRBL itself? First I had NO wiring, changed to NC (and of course changed GRBL configuration - inverted limit pins), but it doesn't help. Also I have very long homing switch debounce time, 250 ms, but does this setting work only in homing mode or always for limit switches?

GrahamJB, 2023-07-18:
Perfectly normal behaviour, when a switch is triggered and the router stops the switch will still be triggered. When you move away from it the axis will move but the  switch is still triggered so another alarm is generated.

 

It can take 2 or 3 tries to jog away far enough before the switch un-triggers, how many tries depends on a load of things, mostly mechanical, not software related.

GrahamJB, 2023-07-19:
> How can I prevent this from happening? I want the limit switch error to appear only when it's pressed, not released. Does GRBL interpret every change in input state as limit switch error, disregarding whether it's change from 0 to 1 or vice versa? Or can it be fixed with some low-pass filter, i.e. RC?

It’s mechanical!

The router approaches a limit switch at a certain speed, when it triggers the axis stops.

But nothing stops instantaneously so it has kept moving a certain distance, depends on speed, inertia…., past the trigger point.

Also the trigger point, the position where the switch triggers, is not the same as the ‘un-trigger’ point, the position at which the switch releases.

 

No RC filter or anything else will change this behaviour.

SebastianKusykIFJ, 2023-07-19:
> But nothing stops instantaneously so it has kept moving a certain distance, depends on speed, inertia…., past the trigger point.
It doesn't change anything. After I read the "limit switch" message, click reset, click unlock, send a command to move in the opposite direction, it has more than enough time to stop moving. Even if it were moving, it wouldn't affect limit switch's state (of course unless it's crushed) - it can be either pressed or not, nothing else.

> Also the trigger point, the position where the switch triggers, is not the same as the ‘un-trigger’ point, the position at which the switch releases.

I know the 'trigger' point is a bit deeper than 'un-trigger', but again: the switch is still pressed when I reset GRBL, then I start moving in the opposite direction and I release the limit switch - so why GRBL behaves like I pressed it, not released?

lalo-uy, 2023-07-19:
> Because when it starts to move it is still pressed, if you look at the
> wiring the limit switches at each end of an axis are connected to the same
> Arduino Pin. Grbl doesn’t know which one is triggered, just that one of
> them is.
>
>
>
> Grbl is reacting to a switch being triggered, , still triggered, left or
> right one, it just does not know.

The limit switches are level trigered, not edge.

SebastianKusykIFJ, 2023-07-20:
Which parts of source code are responsible for limit switches? I tried to find it, but I couldn't.

mar0x, 2023-07-20:
[grbl/limits.c](https://github.com/gnea/grbl/blob/master/grbl/limits.c)