gnea/grbl #1224: 3018 z axis Inverted

dfkchase, 2023-07-28:
So I installed some homing switched and when I did the set up for it I had to invert the Z axis so it would home up instead of down, but then even just jogging the machine around the z axis is inverted and cant seem to fixe it. I thought Having limit switches was going to make things better not worse.  I'll leave my settings below. When z homes up as it should $3 = 2. It's currently crashing into the bed again with the below settings

Anyways, any help is appreciated. 

$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$3 = 4    (Step direction invert, mask)
$2 = 0    (Step pulse invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$6 = 0    (Invert probe pin, boolean)
$12 = 0.002    (Arc tolerance, millimeters)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$13 = 0    (Report in inches, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 5    (Homing direction invert, mask)
$25 = 500.000    (Homing search seek rate, mm/min)
$24 = 25.000    (Homing locate feed rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$30 = 0    (Maximum spindle speed, RPM)
$27 = 1.500    (Homing switch pull-off distance, millimeters)
$31 = 0   (Minimum spindle speed, RPM)
$100 = 800.000    (X-axis travel resolution, step/mm)
$32 = 0    (Laser-mode enable, boolean)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 1000.000    (X-axis maximum rate, mm/min)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$112 = 600.000    (Z-axis maximum rate, mm/min)
$121 = 30.000    (Y-axis acceleration, mm/sec^2)
$120 = 30.000    (X-axis acceleration, mm/sec^2)
$122 = 30.000    (Z-axis acceleration, mm/sec^2)
$130 = 290.000    (X-axis maximum travel, millimeters)
$131 = 170.000    (Y-axis maximum travel, millimeters)
$132 = 43.000    (Z-axis maximum travel, millimeters)

neilferreri, 2023-07-28:
Changing $3 from 2 to 4 and back, changes both the Y & Z axis direction. Get your $3 correct for all axes to move in the correct direction when jogging.
If the Z moves down during homing, adjust your $23 value to correct it. ($23=1)

dfkchase, 2023-07-28:
> Changing $3 from 2 to 4 and back, changes both the Y & Z axis direction. Get your $3 correct for all axes to move in the correct direction when jogging. If the Z moves down during homing, adjust your $23 value to correct it. ($23=1)

Ah, thank you!!! I Had to change $3=6 before $23=1 would work, but its a successful homing and jogging scenario! Finally, all axis' work as expected. Muchly appreciated!

closed 2023-07-28 by dfkchase