gnea/grbl #1225: How to disable spindle gradient startup (pwm) ?

ASHIT-AXAR, 2023-08-06:
hello there...
i need pwm but this gradient spindle startup makes trouble for my use.
i need instant pwm output when i send spindle value.
what can i do ? is there a switch in config files ? or do i need to change source codes the hard way?
thank you.

AlexHolden, 2023-09-07:
I believe the change in PWM output is instant (see line 124 of spindle_control.c). If you are seeing the resulting voltage change slowly perhaps your breakout board has an RC filter on the output with a long time constant?

ASHIT-AXAR, 2023-09-12:
hii Alex. yes, actually there is. at first due to the noisy environment i couldn't read the exact PWM signal accurately and i was not sure whether its the GRBL or not. i was in hurry so i just switched to reading it simply by smooth out the PWM to ADC and put another controller in between. later i suspected it is the RC filter. its just working and i decided to leave it as is for now. 
i appreciate for your time and attention.

closed 2023-09-12 by ASHIT-AXAR