gnea/grbl #1236: G19 freezes CPU of : Grbl 1.1f ['$' for help] VIGOTEC.SDEM1.1.0.220328

qbox4u, 2023-11-29:
i have the feeling that this is the right place to address a cnc issue for the Vevor 3018pro cpu firmware.
please correct me if i am wrong.

the issue is quite simple
Fusion 360 generates a perfect output for cnc units, but als used G17, G18, G19
**** CNC part *****
G1 Y15.762
X-28.762
Y-15.762
X27.762
Y-4.174
Y-3.856
G19 G3 Y-3.539 Z-4.682 J0 K0.318
G1 Z5
X21.922 Y-3.161
Z-4
Z-5 F150
G17 G2 X21.617 Y-3.625 I-2.146 J1.078 F500
G1 X21.614 Y-3.629
********************
During execution of some samples, the cnc spindle was keep on running, but no serial communication
was possible anymore. The programm was stuch on the line: G19 G3 Y-3.539 Z-4.682 J0 K0.318
The only way out was to reset the CPU board.

i will show the problem by means of the Candle console 
this shows the communication of the Vevor 3018 pro cnc 
Grbl 1.1f ['$' for help]
VIGOTEC.SDEM1.1.0.220328
GXX < error:2
G545 < error:20
G17 < ok
G18 < ok
G19
GXX
** CPU IS FROZEN**

after reset, the console receives 
Grbl 1.1f ['$' for help]
VIGOTEC.SDEM1.1.0.220328
This is correct

when the 3018Pro receives the false command "Gxx" it will reply "GXX < error:2"
this is correct as error 2 means : “error:2”  : _(“Numeric value format is not valid or missing an expected value.”),
xx is not an numeric value, so this is the correct reply from the 30187

when a wrong G code would be send to the 3018 eg: G545 , the 3018pro replies: G545 < error:20
this would be correct as error 20 means: “error:20” : _(“Unsupported or invalid g-code command found in block.”),

when the 3018pro receives the command G17, it replies : G17 < ok
this is correct

When the 3018 receives the command G19, the 3018 freezes and does not communicate anymore.
this is an design flaw !!!

resuming the above it is proven that the 3018Pro CPU firmware has a design flaw as it does not return 
as expected if the code was not implemented with an error 20, but stops functioning !!!!!

i expect that the SW release of the 3018pro board is: Grbl 1.1f 
VIGOTEC.SDEM1.1.0.22032

breiler, 2023-11-29:
You should probably contact the company that you bought the hardware from.

I am using the default GRBL 1.1f build from here: https://github.com/gnea/grbl/releases/tag/v1.1f.20170801

Output is from UGS, so it is a bit more verbose - but as you can see it works fine:
```
>>> GXX
An error was detected while sending 'GXX': (error:2) Missing the expected G-code word value or numeric value format is not valid. Streaming has been paused.
>>> G545
An error was detected while sending 'G545': (error:20) Unsupported or invalid g-code command found in block. Streaming has been paused.
>>> G17
ok
>>> G18
ok
>>> G19
ok
>>> GXX
An error was detected while sending 'GXX': (error:2) Missing the expected G-code word value or numeric value format is not valid. Streaming has been paused.
```

qbox4u, 2023-11-29:
Thanks Joacim

This confirms for me that the VEVOR 3018Pro indeed should respond to G19
correctly and
has an issue. ( i start to see more people with apparently the same issue
when using fusion 360 )

You were a great help..

qbox4u, 2023-12-01:
VEVOR confirmed that The supplier of the board  did not implement and test the command G19 correctly