gnea/grbl #1239: Connect GRBL cnc to LMDE via USB

petelodge89, 2023-12-30:
I am trying to connect LMDE to my GRBL arduino based cnc via usb, I want to install LMDE on my old mac to control the cnc via UGS Platform I am getting an error message when I try and connect error opening connection. could not connect to controller on port jserial.com://ACMO:115200. I have looked at posts elsewhere and I have tried some suggested  terminal commands which showed error messages ie the commands were not correct for LMDE ie sudo chmod a+rw/dev/ttyUSBO ! Can anyone suggest how I might get the comms to work over USB. Many thanks

Schildkroet, 2023-12-30:
Make sure you have sufficient rights
https://unix.stackexchange.com/questions/14354/read-write-to-a-serial-port-without-root

petelodge89, 2023-12-31:
I tried sudo usermod -a- G dialout $user but that did not help. All I want to do is connect my cnc arduino controller to UGS platform, then I will do a full install wiping out mac os. All I use this imac for is the cnc. I have zero technical knowledge of unix, I don't really need it, I hope for this one task ! cheers, Pete

breiler, 2023-12-31:
Are you certain that your Arduino is properly flashed with GRBL? Using the Arduino IDE, have you tried connecting to the device from the serial monitor?

petelodge89, 2023-12-31:
The system works fine under the older version of mac os running on this imac connected via usb, so I had hoped that LMDE would just connect the same way with the existing usb connection. Why would it do anything else, USB is USB

breiler, 2023-12-31:
Disconnect your Arduino and connect it again then run this `sudo dmesg` to make sure that the device you are connecting to is properly identified.

It should output something like this: 
```
[  181.711150] usb 1-9.2: new full-speed USB device number 15 using xhci_hcd
[  181.829907] usb 1-9.2: New USB device found, idVendor=2a03, idProduct=0043, bcdDevice= 0.01
[  181.829924] usb 1-9.2: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[  181.829930] usb 1-9.2: Product: Arduino Uno
[  181.829935] usb 1-9.2: Manufacturer: Arduino Srl            
[  181.829939] usb 1-9.2: SerialNumber: 85435333231351402161
[  181.854519] cdc_acm 1-9.2:1.0: ttyACM0: USB ACM device
[  181.854583] usbcore: registered new interface driver cdc_acm
[  181.854587] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

Then run this to make sure it has the right group permissions `ls -al /dev/ttyACM0`. Should output something like:
```
crw-rw----+ 1 root dialout 166, 0 dec 31 12:44 /dev/ttyACM0
```

Then run this command to make sure you have the proper group rights `groups`. This should include the **dialout** group.
 
Make sure you don't have `brltty` install, read this for more info:
https://github.com/arduino/help-center-content/issues/155

petelodge89, 2024-01-03:
Hi Joacim, thank you very much for your help. I have run sudo dmesg and it did indeed output the strings.as you show. I then ran the -al command it ouputted the string as you show but the /dev/tty/ACM0 is in yellow. You then ask me to run a command to determine if I have the correct group right, but no command is shown. I am working off an iso on a memory stick, I have checked and I can't see Britty. Many thanks, Peter

breiler, 2024-01-03:
> You then ask me to run a command to determine if I have the correct group right, but no command is shown.

What is the output when you run `groups` from the terminal?

> I have checked and I can't see Britty.

How did you check? Did you run `ps -C brltty`?

petelodge89, 2024-01-03:
when I run groups, I get mint cdrom sudo audio dip video plugdev users netdev bluetooth scanner, when I run sudo ps -C brltty I get PID TTY   TIME CMD

breiler, 2024-01-03:
> I then ran the -al command it ouputted the string as you show but the /dev/tty/ACM0 is in yellow.

Since you didn't give me the actual output I will have to assume it has the group rights for the `dialout` group which is not in your list of assigned groups: `mint cdrom sudo audio dip video plugdev users netdev bluetooth scanner`

This means that you didn't successfully add your user to the group, normally you need to logout/login for it to take effect in the whole user session:
`usermod -a -G dialout $USER`

> I am working off an iso on a memory stick

This is probably a bad idea as the group settings you make are not persisted.