gnea/grbl #1241: alarm 2 soft limit

fre010, 2024-01-09:
Hello,

For the first time I enabled $20=1. 
After Homing, my machine is in X=1.5, Y=-1.5 and Z=-1.5. the X axis can travel in + and Y and Z in -.
$130 is 1000.0
$131 is 350.0
$132 is 230.0
After homing i tried G01 X100 F1000 but nothing happens and it goes into ALARM 2.
Any idea what I did wrong or what is the cause?
it works again when I disable $20.

Kind regards

fre010, 2024-01-09:
$0=10     
$1=25     
$2=0    
$3=5    
$4=0    
$5=1    
$6=0    
$10=1     
$11=0.010         
$12=0.002         
$13=0     
$20=1     
$21=0     
$22=1     
$23=1     
$24=100.000           
$25=500.000           
$26=250       
$27=1.500         
$30=1000        
$31=0     
$32=0     
$100=64.000           
$101=80.000           
$102=160.000            
$110=10000.000              
$111=5000.000             
$112=5000.000             
$120=200.000
$121=100.000
$122=100.000
$130=1000.000
$131=350.000
$132=230.000

these are my parameters

breiler, 2024-01-09:
The [Alarm 2](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/doc/csv/alarm_codes_en_US.csv#L3) means that you have exceeded the soft limits.

Your command `G01 X100 F1000` makes a lot of assumptions of the state of your machine:
* What is the current state of your machine ([$G](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#g---view-gcode-parser-state))?
  * What units are you expecting it use (G20/G21)? 
  * Should the movement be absolute or incremental (G90/G91)? 
  * What woork coordinate system are you using (G54-G59)? 
* What work offsets have you made ([$#](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#---view-gcode-parameters))?
* What is the current position of the machine ([?](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#ascii-realtime-command-descriptions))

fre010, 2024-01-09:
> The [Alarm 2](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/doc/csv/alarm_codes_en_US.csv#L3) means that you have exceeded the soft limits.
>
> Your command `G01 X100 [...]

[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
right now there have not been any work offsets made. so basically G54 is [G54:0.000,0.000,0.000]
this is the current position: <Idle|MPos:1.500,-1.500,-1.500|FS:0,0|Pn:PY|WCO:0.000,0.000,0.000>

breiler, 2024-01-09:
Well, with those states I think that it should be possible for the machine to move to the absolute position X100. You will have to keep experimenting, I have no idea why it is failing.

I noticed that you Y-limit switch is reporting as triggered `Pn:PY`.

fre010, 2024-01-09:
> Well, with those states I think that it should be possible for the machine to move to the absolute position X100. You will have to keep experimenting, I have no idea why it is failing.
> 
> I noticed that you Y-limit switch is reporting as triggered `Pn:PY`.

thanks for the help, i looked at the Y-limit switch, it is not triggered now but still no movement. Weird as I think I did everything as it should be done

breiler, 2024-01-09:
Just to make sure, after you've homed your machine and standing in front of your machine. Is your spindle positioned to the left?

The machine position for X is wrong here, I think that it should be `-998.5`:
`<Idle|MPos:1.500,-1.500,-1.500|FS:0,0|Pn:PY|WCO:0.000,0.000,0.000>`

Can you post the output from `$I`.

fre010, 2024-01-09:
hi, yes, the spindle is left, up and at the back.
this is after homing: <Idle|MPos:1.500,-1.500,-1.500|FS:0,0|Pn:PY|WCO:0.000,0.000,0.000>
this is the $I:
[VER:1.1g.20180813.Mega:]
[OPT:VNM+,35,255]

breiler, 2024-01-09:
> this is the $I:
[VER:1.1g.20180813.Mega:]

Good, I just wanted to make sure you didn't compile it to work in positive space (HOMING_FORCE_SET_ORIGIN).

> how can I edit it so it is at -998.5 and not +1.5?

You shouldn't have to, at least with my test machine it will automatically apply the soft limits when inverting the any homing axis. 🤷🏻

breiler, 2024-01-09:
Do you have a stable USB connection to the controller? No disconnects after homing?

fre010, 2024-01-09:
i think i foud the problem, it was the loose Y sensor, when homing it kept seeing it and that is the reason it was 1.5 for X and not -998.5.

now I adjusted my G54 so that the homing position (-100,0,0) is G54 0,0,0.
so now I assume that if I work in G54, I can go to max 1000, -350, -250?

closed 2024-01-10 by fre010