gnea/grbl #1245: Grbl V1.1

Charlessgordon, 2024-02-07:
I just loaded version 1.1 and I have two problems, maybe more.
1st in the status box "Unknown"
2nd Machine coordinates first box=0.000, 2nd box 0.000 3rd box 0.000|FS:0
3rd, in the display X-Y and where the tool is located never moves, and the X,Y,Z coordinates stay at zero.
The work piece is displayed
Control boxes 4 and 4, first four boxes=1st two boxes are dark last two are not, 2nd, second 4 boxes the last 2 are dark.
seems to be running the mill.
I don't have limit switches.

HELP  please

breiler, 2024-02-08:
What software are you using to connect to GRBL?

Charlessgordon, 2024-02-08:
using Arduino 2.3.0 / windows 10 / CNC 3018 Pro / Fusion 360 / CH341SER /grbl-1.1h20190825
Grbl 0.9 was installed originally when received but had a few bugs ie $ locations 26 thru 32 were not listed.
I downloaded the Grbl 0.9 version and reinstalled it $ locations 26 thru 32 still not listed. 
I've downloaded Grbl V 1.1  thinking corrupt file, same results.
PS the software that came with the unit from China was bad and had a virus.
Maybe a defective board ?
[image]
[image]

I apricate any help you can offer, O- I already have Dynamite, I've been trying to draw a Valentine heart and get this working for two months, so far, I have learned a little about Fusion 360 and have post process and save the file and loaded them in to the CNC .
So far it's been a waste of time and money, but it looks pretty, LOL.

Charlessgordon, 2024-02-08:
I'm using Arduino 2.3.0 to program the Circuit board (Nano per Minh's Man Cave).
Candle Grbl V 1.1 from my desk top computer.
I used Fusion 360 to make the drawing and Post Process
Status box "Unknown" with the V 1.1 and the orange spindle marker doesn't move, but the X,Y,Z moves mechanically but that's without  any material or bits installed, so it may be working alright but not indicating that.

Charlessgordon, 2024-02-08:
I watched Minh's man cave on you tube to assemble and how to upgrade.
He demonstrates how to change the settings on the Candle by changing the $30 to 10,000 spindle speed max using version 0.9.
I can reload version 0.9 and see if that works better, but everybody else seems to be able to make the V1.1 work.
WOW what a can of worms, just frustrated, but glad you have taken time to help!

Charlessgordon, 2024-02-08:
the Link you listed shows the same board that I have.
Thanks for the info

breiler, 2024-02-09:
What is the output if you connect to the controller via the Arduino IDE:s Serial monitor?

Charlessgordon, 2024-02-09:
in the link you sent for the Amazon board this is listed 
"It is not recommended to update the software or firmware of the control board, otherwise the control board will be damage.
Do not support win10/win8/Mac OS system. The compatibility of the win10 system is very poor, the GRBL control board and the window10 system may not be compatible, and unknown (unsolvable) problems may occur. In order to ensure a good shopping experience for buyers, we have made a reminder on the product page that doesn't support win10 system."

I use Candle to run the CNC not Arduino, I used Arduino just to program the Nano board.

when I used the Arduino tab of tools Serial monito i got the following msg,
"Message (Enter to send message to 'Arduino Nano' on 'COM6')
New Line
115200 baud

Grbl 1.1h ['$' for help]"

I ordered the Amazon board to see if it will work, if not I'll return it.

Any idea why with Candle and grbl V 0.9 the Staus box works but V1.1 says "Unkown" and the G-code panel (which shows the tool and XYZ coords remain at ) or tool doesn't move ?
Isaw somewhere where they had some pin out reversals (345 and 678 as I remember, were swapped)

I did try my laptop windows 10 also same results.

I just want to thank you for trying to help.
I'll try and contact the Chinese company and see if they will exchange the circuit board, they said there were some bad boards before and would exchange the board bet then they went silent, no surprise there .

This Grbl v1.1 has been out for year and nobody has had the problems I have had, must just be me.
Thanks again!!!

closed 2024-02-09 by Charlessgordon

Charlessgordon, 2024-02-09:
> GRBL 0.9 is the old version, which does not have those settings:
> https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#grbl-settings
> Depending how they have compiled the firmware it may or may not be directly compatible with GRBL 1.1 which has changed the pin mapping.
> GRBL 0.9 and older: https://github.com/grbl/grbl/wiki/Connecting-Grbl
> GRBL 1.1 and newer: https://github.com/gnea/grbl/wiki/Connecting-Grbl
> The product page for a similar control boards indicates that it is compiled using the latest pin mapping since they claim that it supports variable spindle. I have also seen this board being marketed with GRBL 1.1. So it might be compatible with GRBL 1.1 out of the box 🤷🏻‍♂️:
> https://www.amazon.com/RATTMMOTOR-Control-Engrving-Machine-Controller/dp/B08BNVBLJR/ref=sr_1_1_sspa?crid=318ODAGBHZ6IH&keywords=RATTMMOTOR%2BGRBL%2B1.1%2BRed%2BControl%2BBoard&qid=1707412730&sprefix=rattmmotor%2Bgrbl%2B1.1%2Bred%2Bcontrol%2Bboard%2Caps%2C156&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1
> How are you connecting to the software: UGS, Candle, Arduino IDE?
> What is the output/response from the controller in the console?

Candle,Sent from my iPhoneCharles S Gordon

Charlessgordon, 2024-02-09:
I am using Arduino 2.3.0 to program the Nano board. and Candle Grbl V1.1
I used Fusion 360 to draw and Post Process.
The CNC moves as though it cutting, but I don't have a bit in it,(trying it
in my office before going to the shop)
the Display panel(G-code program) does't move(tool icon nor the cords don't
move and stay at 0.000.
The Status box="Unknown" console box ="[CTRL+X] < Grbl 1.1h ['$' for help]"

breiler, 2024-02-09:
> I use Candle to run the CNC not Arduino, I used Arduino just to program the Nano board.

I wanted you to connect with the Arduino IDE to see if GRBL got properly installed and that you could initiate a connection to it. And since you got the `Grbl 1.1h ['$' for help]` response is a sign that you have been able to install the firmware. 

The question is if the pin mapping is correct. As I couldn't find any documentation for the controller board your guess is as good as mine - you will simply have to try. But in order to do that you need a working gcode sender.

> Any idea why with Candle and grbl V 0.9 the Staus box works but V1.1 says "Unkown" and the G-code panel (which shows the tool and XYZ coords remain at ) or tool doesn't move ?

I develop on UGS and have never run Candle so I can't say why Candle reports status "Unknown". But it is a sign that the version of Candle you are running is not able to properly connect or recognize the controller. 

There are other senders that you can try: UGS, gSender, CNCjs, bCNC.