gnea/grbl #1246: Different ports for Step and dir pins of each stepper motor

TITAS1533, 2024-02-22:
Hello,
Recently started building cnc machine and for the brains I choosen the old creality 1.1.4 board With atmega1248p ic on it but the problem is how to make all steppers work because all step and dir pins are on diferent ports (picture). Is there any workaround?
[image]

breiler, 2024-02-22:
This flavor of GRBL does not support Atmega1248p. You should try and find one that does.
A quick google gave this: https://github.com/vectorcrumb/grbl-Melzi-3.5X

TITAS1533, 2024-02-22:
officialy no but still managed to make it run the code ,but is there a way to make grbl run step and dir pins on different ports for each stepper motor driver?

breiler, 2024-02-22:
Here is how the fork handles the pin mappings:
https://github.com/vectorcrumb/grbl-Melzi-3.5X/blob/main/GRBL-Mega-5X/src/cpu_map.h#L428