gnea/grbl #1248: stream.py for python3

dlbraun, 2024-03-10:
I did a first pass at updating doc/scripts/stream.py for Python3. If one of the developers wants to check it over and add it to the distribution that would be great.   It basically works.  There is an issue where after uploading all the lines from the g-code file it sometimes enters an infinite loop of printing IDLE:.... 

[stream.txt](https://github.com/gnea/grbl/files/14550747/stream.txt)

Mitko125, 2024-10-05:
What is the possibility of getting your "stream.py"? I'm not experienced in python and I'm having trouble converting the original to python3.

dlbraun, 2024-10-05:
Here it is.

I haven't looked at it in a while.

Best,

David

Mitko125, 2024-10-05:
Thank you.
But I can't figure out how to download "stream.py" at all?

dlbraun, 2024-10-05:
#!/usr/bin/env python
"""\

Stream g-code to grbl controller

This script differs from the simple_stream.py script by
tracking the number of characters in grbl's serial read
buffer. This allows grbl to fetch the next line directly
from the serial buffer and does not have to wait for a
response from the computer. This effectively adds another
buffer layer to prevent buffer starvation.

CHANGELOG:
- 20170531: Status report feedback at 1.0 second intervals.
   Configurable baudrate and report intervals. Bug fixes.
- 20161212: Added push message feedback for simple streaming
- 20140714: Updated baud rate to 115200. Added a settings
 write mode via simple streaming method. MIT-licensed.

TODO:
- Add realtime control commands during streaming.

---------------------
The MIT License (MIT)

Copyright (c) 2012-2017 Sungeun K. Jeon

Permission is hereby granted, free of charge, to any person obtaining a
copy
of this software and associated documentation files (the "Software"), to
deal
in the Software without restriction, including without limitation the
rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
---------------------
"""

import serial
import re
import time
import sys
import argparse
import threading

RX_BUFFER_SIZE = 128
BAUD_RATE = 115200
ENABLE_STATUS_REPORTS = True
REPORT_INTERVAL = 1.0 # seconds

is_run = True # Controls query timer

# Define command line argument interface
parser = argparse.ArgumentParser(description='Stream g-code file to grbl.
(pySerial and argparse libraries required)')
parser.add_argument('gcode_file', type=argparse.FileType('r'),
       help='g-code filename to be streamed')
[... 128 lines cut ...]
           if out_temp.find('error') >= 0 : error_count += 1
           g_count += 1 # Iterate g-code counter
           del c_line[0] # Delete the block character count corresponding
to the last 'ok'
           if verbose: print "  REC<"+str(g_count)+": \""+out_temp + "\""

# Wait for user input after streaming is completed
print "\nG-code streaming finished!"
end_time = time.time();
is_run = False;
print " Time elapsed: ",end_time-start_time,"\n"
if check_mode :
   if error_count > 0 :
       print "CHECK FAILED:",error_count,"errors found! See output for
details.\n"
   else :
       print "CHECK PASSED: No errors found in g-code program.\n"
else :
  print "WARNING: Wait until Grbl completes buffered g-code blocks before
exiting."
  raw_input("  Press <Enter> to exit and disable Grbl.")

# Close file and serial port
f.close()
s.close()

Mitko125, 2024-10-06:
Unfortunately, as you sent me the file, I cannot insert it directly into "VS Code". I started to redo it and noticed that expressions like:
print "Initia.....
are for python2 and not for python3 which wants:
print("Initia...

So I don't need it, but thank you for the effort.