gnea/grbl #1254: Homing on x y z at the same time

Ninobr, 2024-07-12:
Hello everyone
I'm creating a project with 3 stepper motors, controlled by Arduino with GRBL 1.1h installed, the motors move 3 micrometric screws. 
The 3 screws are vertical pointing up and they hold an aluminum plate which must work horizontal. 
The system works and now I added the limit switches to always start from the same position also to calibrate the horizontal alignment with a spirit level in order to correct for different height of the switches after homing. 
I tried to enable the homing procedure but the sequence of motors in cycle 0 is the homing of the z axis, then in cycle 1 homing of the x and y axes. 
This is correct for CNCs, but, in my case, I would need homing in cycle 0 of all the x, y, and z axes at the same time to avoid tilting of the plate during homing. 
I didn't find information on the web and I tried to modify the GRBL config.h file by putting all the axes together,
#define HOMING_CYCLE_0 ((1<<Z_AXIS)|(1<<X_AXIS)|(1<<Y_AXIS))
but compiling GRBL gives an error, since this is an unusual case not foreseen.
I hope someone can help me solve this problem, thanks in advance
Nino

jeffreyb11, 2025-01-12:
just thinking out loud here.
 can you split the wires to the stepper drivers and use only 1 channel?  then it will all run simultanous, but wont level automaticly,
at least its homing then,  leveling can then be done by hand and when it doesnt crash it should stay level

neilferreri, 2025-01-12:
`#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)|(1<<Z_AXIS)) `
I think this should work as expected.  
Make sure you comment out` //#define HOMING_CYCLE_1 `

#compiling GRBL gives an error, since this is an unusual case not foreseen
What error?

Ninobr, 2025-01-19:
Many thanks jeffreyb11 and neilferreri for your suggestions.
I need to level the system, which is part of an atomic force microscope setup. The system operates in microstepping mode, with steps of approximately 0.1 microns (using micrometric screws that have 250 microns per turn). The system always starts at the limit switch position for exchanging samples and also to ensure accuracy. In fact I must be certain that the system returns to the same position each time, in case microsteps are lost, since I am using also single microsteps, but there is no guarantee that all motors will execute them correctly. Therefore, I reset the system every time to maintain consistency.
At the moment I am busy on another work for a week or so, but for sure I will try 
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)|(1<<Z_AXIS)) 
and I will post about the results