gnea/grbl #1256: $30, $31, $32 setup for laser

Marmadoc, 2024-09-09:
I use GRBL 1.1 with laser cutter.
I do not understand $30, $31 and $32
$31 and $32 is almost understandable, but $30 is not clear.
Why my laser is working with 100% all time when $32?
how exactly $30 number is related to PWM of laser.
Some suggestion $32 to 0, some to 1, some to 255 or even 1000.
Please help me to understand this.
Thx in advance

breiler, 2024-09-11:
And you have read this?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm

Marmadoc, 2024-09-11:
Of course I read it - and I try to understand it in relation to laser, not spindle.
$30 - "if you want to set 10000rpm at 5V, program $30=10000. For 255rpm at 5V, program $30=255"
          But how it works if I need 100% laser power - I understand at 5V - ? 
         is 1000rpm  the same as 100% power?
$31 - if set to 0 - it is 0% laser power?
$32 - My Gcode use M3 and M5 - I understand it is M3-laser ON, M5 - laser OFF. If laser is ON with power set by software 50% - how it is translates to set up $30.
I'm very sorry if my questions are unprofessional, but I'm not programmer.
Thx in adavnce

Marmadoc, 2024-09-11:
Hi Farukguelle - could you please contact me by email aszczurowski@hotmail.com - Thank you

Farukguelle, 2024-09-12:
Sure, I am glad to help. May I ask why the contact via Mail? In case you have open questions let me know and I'd be glad to help here :] 

Also one more information that can't be told enough is to handle the Laser with caution. I dont know how much your Laser's output is but since it must be enough to engrave I assume it to be dangerous enough to damage the eyes if not taken proper care. If you want to test the Intensity of the Laser make sure you have it pointed to something non reflecting, not burning down or causing a fire, etc... Also always wear Laser goggles of the OD (Optical Density) specific for your Laser. I personally wear OD4+ glasses, besides from already 2 protective glasses surrounding the Laser head. I only have one pair of eyes and its just a matter of seconds if something goes wrong so please be safe.

Marmadoc, 2024-09-14:
Thx for info. I thought that some mine questions are not exactly related to GRBL, so I wasn't sure if this is right place to ask. Actually my biggest problem I already solved. After your explanation - finally I understood $30 setup, but my laser didn't go for 100%. I changed to GRBL laser and it was working fine, but not in  Lightburn.  I find out that Lightburn has its own setup not tighten to GRBL $30. The default setup for laser max power to 255. So if grbl was set to 100, 100% in Lightburn was actually 1/4 of full power. Now I have all OK.
I like to ask question about grbl controller. I have CNC arduino shield. On other group I have been told to change for belter (32bit) controller, but I do not know why.
The last think - glasses - I do not have one, but my laser has protective extension to almost working surface.
Thx again a lot.

Farukguelle, 2024-09-14:
That's great to hear that I was able to help you guiding towards the right troubleshooting. As for the max Laser Power in Lightburn: Yeah its a setting spmewhere in the settings of lightburn but it seems like you solved it :] 
Regarding the 32 Bit Controller, I wouldn't know why, since for the basic laser engraving you won't need anything more. 

I personally run it on a self designed Mainboard PCB with the same controller as the Arduino and some silent stepper drivers. I once was planning on writing some code to make it read G Codes from a SD card and show me the result on a Display but I am already working on some projects which made me focus on them rather than the already functioning Laser.

Marmadoc, 2024-09-20:
Thank You. I didn't close this because I have one more problem to understand.
As laser - laser is connected to Z axis and $32 = 1. 
If it is CNC Z axis works Up/Down and spindle is attached to spindle pins on CNC shield.
I use my machine also as a plotter. In this case Z axis works as pen Up/Down but I need modified version of GRBL (for z axis)
How it all is related to PWM? Can I use regular GRBL and control servo or stepper by Z axis?
Can you help me in this matter?

Farukguelle, 2024-09-20:
If I understand it correctly you want to add a servo motor in order to change the Knife's direction since the sharp edge is on one side, is that the issue? 
If so you'd need to replace a digital pin with a function that changes it's value depending on the DIR and STEP pins for the x and y axis (Digital Pin 5 and 6). The function is really dependant on your hardware design though as it will also decide which values to choose for having the knife straight to the front or any other direction.

Marmadoc, 2024-09-20:
No exactly. It is not for knife. It is for pen plotter. Pen plotter must lift the pen and lower the pen. Mechanically it is done by Stepper motor or servo motor. If the stepper motor is used it is controlled by Z driver on CNC shield. In case of the servo motor - it is controlled by Z+ axis (on CNC shield)  - but in this case modified GRBL is used. I know it is related to PWM - but do not know how. What is different between PWM used by Z+ for laser and PWM used by Z+ for servo motor.

lalo-uy, 2024-09-20:
Laser PWM goes fron 0 to 100%
Servo motors usea narrow and time specific range.

Marmadoc, 2024-09-23:
If laser is connected to Z+ on CNC shield - it is adjusted by voltage. 0% = 0V 100%=12V (laser is 12V) $32=1
So in this case it is nothing with PWM.
If Servo motor is connected to Z+ $32=0 this GRBL 1.1 doesn't work. 
I think it is no other way but use modified GRBL software - or instead of servo use Stepper motor

lalo-uy, 2024-09-23:
You can setup for pwm and modify G-code to send M3 xxx for PenUp and M3 yyy
for PenDown. Just find xxx and yyy for the desired angles in the servo.
Need PWM freq of 50hz aprox.

Marmadoc, 2024-10-06:
I'm still learning and trying. Now I like to ask about $31 - minimum S value for minimum (Lightburn). I read that it could be 0 or 5.  I'm not sure if I'm right when I set it to 5 for my laser/engraver.
Again thx in advance