gnea/grbl #1257: Distance correct, but wrong speed

SebastianKusykIFJ, 2024-09-25:
Hello, I have a problem with my XY machine. Maximum speed is set to 500 mm/min (if that is speed unit in GRBL) in both axes. My drives have 1.8 deg/step motors and 8 mm/revolution lead screws. When I tell it to go for some distance, it does it correctly, however it's speed is higher than should be according to this parameter. Here is a table containing the distance (movement in 1 axis only) and measured time it takes to complete:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:\Users\Legion\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\Legion\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

s [cm] | t [s]
-- | --
0.2 | 0.8
0.3 | 0.9
0.4 | 1.0
0.5 | 1.1
0.8 | 1.4
1 | 1.5
2 | 2.0
3 | 2.5
5 | 3.4
8 | 4.9
10 | 5.8
12 | 6.8
15 | 8.2
20 | 10.5
30 | 15.3

</body>

</html>
It means that the real maximum speed is about 2 cm/s, while 500 mm/min is 0,83 cm/s. It must be a problem with GRBL, not stepper drivers, because I connected an oscilloscope to the PUL output and the frequency of pulses sent by GRBL is 2.080KHz - for 4 pulses per cycle and 1.8 deg motor that's 2.6 revolutions/s, then the 8 mm/revolution screw converts it to 2.08 cm/s linear motion. How can I make the maximum speed set in GRBL match the real one?

breiler, 2024-09-25:
Not sure what you are asking, the maximum feed rate is set per axis.

If you move both X and Y at the same time your maximum feed is the hypotenuse of the both axes, which should be about 707 mm/min or 12mm/s: https://www.calculator.net/right-triangle-calculator.html?av=500&alphav=&alphaunit=d&bv=500&betav=&betaunit=d&cv=&hv=&areav=&perimeterv=&x=Calculate

I am guessing that you are sending G0 commands which will move the machine as fast as possible. If you want a specific feed rate you should use G1 (together with the appropriate feed rate mode https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g93-g94-g95)

SebastianKusykIFJ, 2024-09-25:
> If you move both X and Y

No, I don't. I'm asking about one axis movement. 

> I am guessing that you are sending G0 commands which will move the machine as fast as possible.

'As fast as possible' shouldn't be more than speed limit set in GRBL parameters, right? But for some reason it is. If I set it to 500 mm/min, my drive should move 0,83 cm/s, while in reality it moves about 2 cm/s. If I set the distance to steps number ratio wrong, distances would also be wrong (different commanded and measured distances), but they are correct - only the speed is wrong.

mar0x, 2024-09-25:
@SebastianKusykIFJ, what is your current [steps/mm setting ($100, $101)](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm) ?

breiler, 2024-09-25:
Sorry, I missed this in your original post `(movement in 1 axis only)`.

> If I set it to 500 mm/min, my drive should move 0,83 cm/s

Generally yes, but you also need to take acceleration and deceleration. But it sounds about right.

> 'As fast as possible' shouldn't be more than speed limit set in GRBL parameters, right? 

Yes, what are your settings?

SebastianKusykIFJ, 2024-09-25:
These are my settings:
$0=10
$1=255
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

SebastianKusykIFJ, 2024-09-25:
I measured the average speed on 20 cm distance also for 100 and 250 values of GRBL speed parameter (low speeds, so it doesn't take much time to accelerate and decelerate). I got these results:
[image]
It means that the measured speed is about 2.3 times higher than the set one.

mar0x, 2024-09-25:
From your information above: $100=250.000 (steps/mm), each step is 1.8 degrees, so 250 * 1.8 = 450 degrees = 1.25 rev => 1.25 * 8 mm/rev = 10 mm.
It doesn't makes sense. Maybe you have microsteps enabled ?
But you also state the positioning is correct. This means $100 setting value is also correct for your CNC.

breiler, 2024-09-25:
Did a quick test on my machine.

Ran `G0 X100` with my machine set to max `200 mm/min`, it took ~30s for it to complete. 
Ran it again with the max `400 mm/min` which took ~15s.
And again with max `800 mm/min` at ~7.5s.

So this is working fine for me

SebastianKusykIFJ, 2024-09-26:
Today I did a test to make sure the PC program I'm using is sending correct GRBL commands (if there were wrong coefficients both in program and GRBL, they could compensate to get the right distance, but velocity could be still wrong). So I set the machine in 2 subsequent points, distance between which is 100 mm in X axis (measured with a ruler). To get to the first point the program sent:
G0 X-9.842Y-23.622
and to the next one:
G0 X-49.213Y-23.622
Units are inches/10, because I don't know a way to make GRBL operate in mm without the need to tell it to do so after each soft reset. So distance between these points, according to GRBL, is 39.371 in/10, which equals 100.00234 mm. So why velocity is higher than limit if distances are correct?

mar0x, 2024-09-26:
[$100 - Max Rate](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#110-111-and-112--xyz-max-rate-mmmin) setting limits the rate on **each axis** independently. Consider to change your experiment to move along one axis.

SebastianKusykIFJ, 2024-09-26:
I wrote 2 times that it moves in ONE axis. You can see that Y coordinate is the same for both points, only X changes.

mar0x, 2024-09-26:
>  .. Y coordinate is the same for both points, only X changes.

Ok. Got it. Sorry about that.

> I don't know a way to make GRBL operate in mm without the need to tell it to do so after each soft reset. 

What is your GRBL version then?

SebastianKusykIFJ, 2024-09-26:
This is VER:1.1h.20190830, OPT:VZ,15,128. I used LaserGRBL to flash that firmware and it was provided by that program - custom, XY homing.

breiler, 2024-09-26:
> Units are inches/10, because I don't know a way to make GRBL operate in mm without the need to tell it to do so after each soft reset. So distance between these points, according to GRBL, is 39.371 in/10, which equals 100.00234 mm. 

What do you mean by this? 

Just to be clear, how far does your machine move if you send this command?
```
G91 G21 G0 X100
```

SebastianKusykIFJ, 2024-09-26:
With that command it moved 250 mm.

breiler, 2024-09-26:
That command should have moved 100 mm, so your settings are wrong.

You have configured your controller with "steps/inch" instead of "steps/mm".

Set your steps settings to `$100=100` and send gcode commands in metric units (G21) and it should work.

breiler, 2024-09-30:
Did you get it working?

SebastianKusykIFJ, 2024-10-03:
Thank you, now it works. Actually the command that switches in/10 to mm did nothing, because GRBL already accepted commands in mm. We thought that it accepts commands in in/10, because 250/100=2.5, which is almost 2.54. However, when we measured positioning more accurately, there were small errors for long distances because of that 0.04. We set $100 and $101 to 100, changed the PC program so it doesn't convert cm to in/10, but to mm, and now it works correctly.

closed 2024-10-03 by SebastianKusykIFJ