gnea/grbl #1258: Can anyone help me

pure08simple, 2024-10-20:
i wish to know how can i pass the error message "machine alarm"

pure08simple, 2024-10-20:
it shows everytime.
i would like to start doing something. i don't know if it's some mal connetion or anything else.

RalDee, 2024-11-05:
you could try "Reset Alarm" and "Machine reset" 
I have this message sometimes at the first power-on after a powerless time. 
If this don't help control the settings of your endswitches.

mar0x, 2024-11-06:
Hello,
There should be numerical alarm code. Using the code and documentation you may try to discover the reason of the alam.

jeffreyb11, 2025-01-12:
some programms need to home to unlock the alarm allso

RoLeGr, 2026-02-06:
>some programms need to home to unlock the alarm allso

That's it I think: if you look at your config.h file, you have the following lines:
```C
// If homing is enabled, homing init lock sets Grbl into an alarm state upon power up. This forces
// the user to perform the homing cycle (or override the locks) before doing anything else. This is
// mainly a safety feature to remind the user to home, since position is unknown to Grbl.
#define HOMING_INIT_LOCK // Comment to disable
```

You can disable this.
An other bypass would be to use the `$X` command when you get the alarm => it will set your current location as your origin