gnea/grbl #1271: SainSmart distributing modified binaries with no attribution nor source code

ericdraken, 2025-02-19:
Maybe this is not an issue for this community, but SainSmart is distributing a compiled version of your GRBL 1.1f (v1.1f.20170801) without the GPLv3 license nor attribution nor source code. Customer support for SainSmart refuses to give the modified source code for GRBL in part (cpu_map.h, config.h) or in whole in order to compile GRBL 1.1h for their Genmitsu CNC hardware. They distribute your works via USB stick and download.

I've had a long email chain with SainSmart, as well as with the FSF (Free Software Foundation) and their GPL-violations team, and they advised me to get the GRBL community in the loop.

I can provide evidence I shared with the FSF if this issue is of merit.

akurpukov, 2025-03-28:
And what, who cares?! We, in 3rd world take it very simple. It is very common to use even cracked software in goverment organizations.

ericdraken, 2025-03-28:
The contributors in this repo whose unsung, unpaid work power the majority of CNC hardware, and all they ask is a company profits from their work attribute them. 

Your “3rd world” “cracked” comment is useless in this context. Read the room and what repo you are in.

Also, the Trade Department would care if the authors were to peruse this as it is a crime. SainSmart is a US company, and money crosses state and country borders.